notro/fbtft #603: ST7789v inverted colors

mountainstorm, 2023-12-03:
Setup:
- Linux raspberrypi 6.1.0-rpi4-rpi-v8 #1 SMP PREEMPT Debian 1:6.1.54-1+rpt2 (2023-10-05) aarch64 GNU/Linux
- Raspberry Pi Zero 2 W
- https://www.aliexpress.us/item/3256804430946996.html (2.4" ST7789v screen)
- Only screen SPI connected (to SPI0, CS0), DC=GPIO24, Reset=GPIO23,LED=GPIO18

I've tried every device overlay and config I can find and the best I have managed to achieve is by using 
`dtoverlay=fbtft,st7789v,reset_pin=23,dc_pin=24,led_pin=18,rotate=270,bgr=1,fps=30,width=320,height=240`

This starts the screen as `/dev/fb0`, I can see the boot messages and seems to work, but the screen is inverted.  The "background" is white (not black) and where `[OK]` should be green it's purple.  Try as I might I can't figure out how to change this.  Note: `bgr` appears to have no effect regardless of what I set it to.

`startx` also just gives be a blank white screen, though I suspect that's related to the fact I only have one `/dev/fb` entry at the moment

notro, 2023-12-03:
You need to find the initialization sequence that matches your display. This is usually found inside the accompanying example code, which seems to missing here. The display is listed as a  MSP2402/MSP2401 and there's a wiki for that: http://www.lcdwiki.com/2.4inch_SPI_Module_ILI9341_SKU:MSP2402
Here it's listed as having a ili9341, but this might have changed in newer batches.

I suggest you contact the seller to get some example code or the init sequence.

Here's the init sequence for the driver you are using: https://elixir.bootlin.com/linux/latest/source/drivers/staging/fbtft/fb_st7789v.c#L143

mountainstorm, 2023-12-03:
Is there an example dtoverlay which "just" sets the init sequence which I can crib from when I figure out what it is?

mountainstorm, 2023-12-04:
oh actually, the overlay here seems to achieve that i.e. everything works except it's inverted.  I tried this hours ago but must have had some other config broken, now I've fixed those problems I tweak the pinout and that works https://github.com/darrenliew96/gamepi20_drivers/blob/main/gamepi20.dts

mountainstorm, 2023-12-04:
haha, victory.  Looking at the data sheet the invert command is 21h and the init sequence there includes 										0x1000021; remove that line 5th line in the init sequence and bingo, colors the right way round.

Uploading here for posterity and anyone in the future with this issue 
[gamepi20.dts.txt](https://github.com/notro/fbtft/files/13543365/gamepi20.dts.txt)

closed 2023-12-04 by mountainstorm