notro/fbtft #604: Column offset

mattfbacon, 2023-12-20:
I bought this display from Adafruit: https://www.adafruit.com/product/5394

The fbtft driver works well for it, except that there is this "colstart" parameter in the example code on Adafruit (e.g., https://learn.adafruit.com/adafruit-1-9-color-ips-tft-display/python-usage). I think it means that the first visible column is actually at column 35.

I observed that when I put the correct width and height in the fbtft parameters, there is 35 pixels of garbage at the bottom and everything is offset by 35 pixels. Adding 35 pixels of width makes the garbage go away but there are still 35 pixels of dead space above the display that are not visible. I can work around this in software by putting stuff below that point, but it's still annoying.

I looked at `/sys/class/graphics/fb0/pan` because it seemed like it might work, but I got "Invalid argument" no matter what I tried to write to it.

Would it be possible to add column offset as a parameter for the module?

PS My dtoverlay config is as follows:

```
dtoverlay=fbtft,spi0-0,st7789v,width=205,height=320,rotate=90,reset_pin=24,dc_pin=25
```

Note 205 = 170 + 35

notro, 2023-12-20:
I haven't done anything but bugfixes on fbtft the past years, so I don't follow where the development is going, but I think if you want that change you need to do it yourself. There's no kernel maintainer for fbtft.

There's a modern version (DRM driver) of fbtft that supports these MIPI DBI panels: https://github.com/notro/panel-mipi-dbi/wiki

It supports setting x/y-offset in the overlay: https://github.com/raspberrypi/linux/blob/rpi-6.1.y/arch/arm/boot/dts/overlays/mipi-dbi-spi-overlay.dts

mattfbacon, 2023-12-20:
I was not aware of panel-mipi-dbi. Thanks, it works perfectly.

[image]

mattfbacon, 2023-12-20:
For future readers, my new `/boot/config.txt` has:

```
#dtoverlay=fbtft,spi0-0,st7789v,width=205,height=320,rotate=270,reset_pin=24,dc_pin=25
dtoverlay=mipi-dbi-spi,spi0-0,speed=48000000
dtparam=compatible=st7789\0panel-mipi-dbi-spi
dtparam=width=320,height=170,width-mm=46,height-mm=25,x-offset=0,y-offset=35
dtparam=reset-gpio=24,dc-gpio=25,write-only
```

I have `st7789.txt` which contains:

```
command 0x11        # 0x1000011
delay 255       # 0x20000ff

# the fbtft driver overwrites this in set_var(), need to pick the value from the driver
# 0x1000036 0xA0

# dts has rotate=90 + bgr: driver converts to:  MADCTL_BGR | MADCTL_MV | MADCTL_MY
command 0x36 0xA8

command 0x3a 0x05     # 0x100003a 0x05
command 0x21        # 0x1000021
command 0x2a 0x00 0x01 0x00 0x3f  # 0x100002a 0x00 0x01 0x00 0x3f
command 0x2b 0x00 0x00 0x00 0xef  # 0x100002b 0x00 0x00 0x00 0xef
command 0xb2 0x0c 0x0c 0x00 0x33 0x33 # 0x10000b2 0x0c 0x0c 0x00 0x33 0x33
command 0xb7 0x35     # 0x10000b7 0x35
command 0xbb 0x1f     # 0x10000bb 0x1f
command 0xc0 0x0c     # 0x10000c0 0x0c
command 0xc2 0x01     # 0x10000c2 0x01
command 0xc3 0x12     # 0x10000c3 0x12
command 0xc4 0x20     # 0x10000c4 0x20
command 0xc6 0x0f     # 0x10000c6 0x0f
command 0xd0 0xa4 0xa1      # 0x10000d0 0xa4 0xa1
command 0xe0 0xd0 0x08 0x11 0x08 0x0C 0x15 0x39 0x33 0x50 0x36 0x13 0x14 0x29 0x2d  # 0x10000e0 [...]
command 0xe1 0xd0 0x08 0x10 0x08 0x06 0x06 0x39 0x44 0x51 0x0b 0x16 0x14 0x2f 0x31  # 0x10000e1 [...]
command 0x29        # 0x1000029
```

I got this from https://forums.raspberrypi.com/viewtopic.php?t=337019

I converted this to `st7789.bin` with https://github.com/notro/panel-mipi-dbi/blob/main/mipi-dbi-cmd then I copied `st7789.bin` to `/lib/firmware/`.

This works perfectly and even uses the proper rotation to make the display run in landscape.

closed 2023-12-20 by mattfbacon